TOLDOTECHNIK/avrdude-gpio-autoreset #1: Getting `stk500_recv(): programmer not responding` on arduino Pro mini 5V

xMTinkerer, 2017-12-17:
Hi!
   I'm trying to program my arduino Pro Mini 5V from a Pi Zero, but I get this stk500 error:

```bash

pi@raspberrypi:~/Arduino/blink/buildstuff $ avrdude -D -V -F -c arduino -p m328p -P /dev/ttyAMA0 -U flash:w:blink.ino.hex:i -vvv

avrdude-original: Version 6.3
                  Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                  Copyright (c) 2007-2014 Joerg Wunsch

                  System wide configuration file is "/etc/avrdude.conf"
                  User configuration file is "/home/pi/.avrduderc"
                  User configuration file does not exist or is not a regular file, skipping

                  Using Port                    : /dev/ttyAMA0
                  Using Programmer              : arduino
Reset sent.strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude-original done.  Thank you.

strace: |avrdude-gpio-autoreset: Broken pipe
```

This is my wiring diagram with the Pi Zero and Arduino v3.3. I've checked my connections over and over, so I'm pretty confident they match the diagram. 

[image]

I've verified I can program the arduino with an FTDI chip to a PC via USB. 

Any ideas? 
I found this, but not sure if it is related:

[DTR auto-reset does not work with Arduino Pro Mini](http://savannah.nongnu.org/patch/?9004)

Random config stuff:

```bash

pi@raspberrypi:~/Arduino/blink/buildstuff $ cat /usr/bin/avrdude
#!/bin/sh
strace -o "|avrdude-gpio-autoreset" -eioctl /usr/bin/avrdude-original $@

pi@raspberrypi:~/Arduino/blink/buildstuff $ uname -a
Linux raspberrypi 4.9.41+ #1023 Tue Aug 8 15:47:12 BST 2017 armv6l GNU/Linux

pi@raspberrypi:~/Arduino/blink/buildstuff $ ll /dev/ttyA*
crw-rw---- 1 root dialout 204, 64 Dec 10 19:30 /dev/ttyAMA0

```

I managed to get my hands on a pro mini 3.3V so I'll give that a shot, but I'm not really expecting anything different.

T94T, 2017-12-17:
Did you do these steps?
`sudo mv /usr/bin/avrdude /usr/bin/avrdude-original`
`sudo nano /usr/bin/avrdude`
paste this code in
`
#!/bin/sh
strace -o "|avrdude-gpio-autoreset" -eioctl /usr/bin/avrdude-original $@
`
and then
`sudo chmod +x /usr/bin/avrdude`

xMTinkerer, 2017-12-18:
Hi. Yea, see above:

```
pi@raspberrypi:~/Arduino/blink/buildstuff $ cat /usr/bin/avrdude
#!/bin/sh
strace -o "|avrdude-gpio-autoreset" -eioctl /usr/bin/avrdude-original $@
```

T94T, 2017-12-18:
Oh, I think you need root permissions to toggle the reset pin. Could you please try running the command with sudo?
`sudo avrdude -D -V -F -c arduino -p m328p -P /dev/ttyAMA0 -U flash:w:blink.ino.hex:i -vvv`

xMTinkerer, 2017-12-18:
Yea, I thought the same thing, but no dice either: (I killed it from the keyboard)

```bash

pi@raspberrypi:~/Arduino/blink/buildstuff $ sudo avrdude -D -V -F -c arduino -p m328p -P /dev/ttyAMA0 -U flash:w:blink.ino.hex:i -vvv

avrdude-original: Version 6.3
                  Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                  Copyright (c) 2007-2014 Joerg Wunsch

                  System wide configuration file is "/etc/avrdude.conf"
                  User configuration file is "/root/.avrduderc"
                  User configuration file does not exist or is not a regular file, skipping

                  Using Port                    : /dev/ttyAMA0
                  Using Programmer              : arduino
Reset sent.strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
^Cstrace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe

pi@raspberrypi:~/Arduino/blink/buildstuff $ 

```

Hmm, I get the same error if I run `avrdude-original`, so I'm wondering if it is even getting far enough?

```
pi@raspberrypi:~/Arduino/blink/buildstuff $ avrdude-original -D -V -F -c arduino -p m328p -P /dev/ttyAMA0 -U flash:w:blink.ino.hex:i -vvv

avrdude-original: Version 6.3
                  Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                  Copyright (c) 2007-2014 Joerg Wunsch

                  System wide configuration file is "/etc/avrdude.conf"
                  User configuration file is "/home/pi/.avrduderc"
                  User configuration file does not exist or is not a regular file, skipping

                  Using Port                    : /dev/ttyAMA0
                  Using Programmer              : arduino
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude-original done.  Thank you.
```

T94T, 2017-12-18:
What does this command show?
`ls /dev/ttyAMA*`
It should give you the ttyAMA0 interface.
If not, try enabling the serial port. raspi-config has some new options. This has to be selected:
[image]

T94T, 2017-12-18:
Don't know the cause. Do you have an oscilloscope to check if the GPIO pin toggles?
I do not have the hardware here to test it right now.

xMTinkerer, 2017-12-18:
Ah interesting. Good call. I'll be traveling for the next week or so, but I can check later. 

I tried holding down the reset button, but that didn't seem to change the output. 

Any other pins I should check?

T94T, 2017-12-19:
I checked the schematics of the pro mini. It seems, RX and TX are labeled the wrong way. I think you just have to swap those two connections. Hope, your raspberry did not get damaged...
http://sfecdn.s3.amazonaws.com/tutorialimages/ProtoSnap_ProMini_Landing/ProtoSnap_ProMini_Schematic_L.PNG

TOLDOTECHNIK, 2018-01-17:
We just released a new version. Could you please try your setup again?

xMTinkerer, 2018-03-03:
Woo! It works. After downloading your new binary, I kept getting this `not in sync: resp=0x48`:

```bash
 
pi@raspberrypi:~/Arduino/blink/buildstuff $ sudo sudo avrdude -D -V -F -c arduino -p m328p -P /dev/ttyAMA0 -U flash:w:blink.ino.hex:i -vvv

avrdude-original: Version 6.3
                  Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                  Copyright (c) 2007-2014 Joerg Wunsch

                  System wide configuration file is "/etc/avrdude.conf"
                  User configuration file is "/root/.avrduderc"
                  User configuration file does not exist or is not a regular file, skipping

                  Using Port                    : /dev/ttyAMA0
                  Using Programmer              : arduino
Reset sent.strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
avrdude-original: stk500_getsync() attempt 1 of 10: not in sync: resp=0x48
avrdude-original: stk500_getsync() attempt 2 of 10: not in sync: resp=0x65
avrdude-original: stk500_getsync() attempt 3 of 10: not in sync: resp=0x6c
avrdude-original: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6c
avrdude-original: stk500_getsync() attempt 5 of 10: not in sync: resp=0x6f
avrdude-original: stk500_getsync() attempt 6 of 10: not in sync: resp=0x0d
avrdude-original: stk500_getsync() attempt 7 of 10: not in sync: resp=0x0a
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 8 of 10: not in sync: resp=0x0a
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 9 of 10: not in sync: resp=0x0a
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 10 of 10: not in sync: resp=0x0a

avrdude-original done.  Thank you.

strace: |avrdude-gpio-autoreset: Broken pipe
```

But after some poking around I tried the `-b 57600` and it works now! So I'm not entirely sure if it was the new binary or actually, explicitly including the baudrate (`-b`) was what did it. 
But either way, I'm a happy camper. 

Thanks for following up!

closed 2018-03-03 by xMTinkerer