TOLDOTECHNIK/avrdude-gpio-autoreset #2: Raspberry Pi 3 strace: |avrdude-gpio-autoreset: Broken pipe

green2blue, 2018-01-16:
I allways getting this Error on Raspberry Pi 3:
strace: |avrdude-gpio-autoreset: Broken pipe

avrdude command is:
sudo avrdude-original -q -q -patmega2560 -cwiring -P/dev/ttyUSB0 -b115200 -D -Uflash:w:/tmp/arduino_build_348361/Blink.ino.hex:i

Any idea what is the Problem?

Kind Regards

TOLDOTECHNIK, 2018-01-16:
What version of Raspbian do you use? We will test it tomorrow.

green2blue, 2018-01-17:
Thank you for your reply.
I use the newest Version:

Raspbian Stretch with desktop
Image with desktop based on Debian Stretch
Version: November 2017
Release date: 2017-11-29
Kernel version: 4.9

TOLDOTECHNIK, 2018-01-17:
There seems to be a problem with avrdude or stretch. Don't know.
If you do a verbose on avrdude-original, it will hang. You can test it by calling:

`avrdude-original -vvvv -D -V -F -c arduino -p m328p -P /dev/ttyAMA0 -U flash:w:PATHTOHEXFILE.hex:i
`

Output will be:

> avrdude-original: Version 6.3
>                   Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>                   Copyright (c) 2007-2014 Joerg Wunsch
> 
>                   System wide configuration file is "/etc/avrdude.conf"
>                   User configuration file is "/root/.avrduderc"
>                   User configuration file does not exist or is not a regular file, skipping
> 
>                   Using Port                    : /dev/ttyAMA0
>                   Using Programmer              : arduino
> avrdude-original: Send: 0 [30]   [20]

Program is never gonna be to finish. We need to have a deeper look into it.

green2blue, 2018-01-17:
When you use the GPIO UART on Raspberry Pi 3. You must call ttyS0 instead of ttyAMA0(thats bluetooth).

OK. Thank you for your help :)

TOLDOTECHNIK, 2018-01-17:
Yes, of course. Didn't think about it anymore. We are going to change the documentation accordingly. Also, we are going to delete some issue comments, so it is more understandable to other users...

TOLDOTECHNIK, 2018-01-17:
Could you please try this?
> sudo avrdude -vvvv -D -V -F -c arduino -p m328p -P /dev/ttyS0 -U flash:PATHTOHEXFILE.hex:i

and  then reset the Arduino manually. Maybe you have to press the reset button multiple times. HEX file is beeing uploaded on our system this way. It seems the reset on GPIO 4 does not get fired automatically. We need to investigate some more time.

green2blue, 2018-01-17:
I use avrdude-original.

Over USB this works: 
sudo avrdude-original -q -q -patmega2560 -cwiring -P/dev/ttyUSB0 -b115200 -D -Uflash:w:/tmp/arduino_build_348361/Blink.ino.hex:i

This with ttyS0 and manually Reset doesn´t work:
pi@HHO-Cell:~ $ sudo avrdude-original -q -q -patmega2560 -cwiring -P/dev/ttyS0 -b115200 -D -Uflash:w:/tmp/arduino_build_348361/Blink.ino.hex:i
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: stk500v2_getsync(): timeout communicating with programmer

And your command:
pi@HHO-Cell:~ $ sudo avrdude-original -vvvv -D -V -F -c arduino -patmega2560 -P /dev/ttyS0 -U flash:/tmp/arduino_build_450883/test.ino.hex:i
avrdude-original: invalid I/O mode '/' in update specification
  allowed values are:
    r = read device
    w = write device
    v = verify device
avrdude-original: error parsing update operation 'flash:/tmp/arduino_build_450883/test.ino.hex:i'
pi@HHO-Cell:~ $

TOLDOTECHNIK, 2018-01-17:
We just encountered two different issues:
- Our GPIO handling routines were not working anymore with Stretch
- The reset capacitor is too small when using an Arduino board. We just tested 1uF (please refer to the new [schematics](https://github.com/TOLDOTECHNIK/avrdude-gpio-autoreset/blob/master/diagrams/RPi%20Arduino%20Wire%20Up.png)). It is strong enough to init a reset.

All new code including the [binary](https://github.com/TOLDOTECHNIK/avrdude-gpio-autoreset/blob/master/bin/avrdude-gpio-autoreset) is now available for download.

This test upload works without problems now. Broken pipe messages can be ignored.

> sudo avrdude -D -V -F -c arduino -p m328p -P /dev/ttyS0 -U flash:w:PATHTOHEXFILE.hex:i

Could you please test again?

green2blue, 2018-01-17:
I changed the two things you write above.
The results is. The Aruduino is rebooting(The LED is blinking). But it doesn´t programm my Arduino MEGA2560.
I always getting this:
pi@HHO-Cell:/usr/bin $ sudo avrdude -D -V -F -c arduino -p m2560 -P /dev/ttyS0 -U flash:w:/tmp/arduino_build_450883/test.ino.hex:i
Reset sent.strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
avrdude-original: stk500_recv(): programmer is not responding
avrdude-original: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
^Cstrace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe

Do you have any hint or ideas that i can bring it to work with the MEGA2560?

After some tests i sniffed the command that the Arduino IDE is sending to avrdude and change it a little bit:
pi@HHO-Cell:/usr/bin $ sudo avrdude -q -q -patmega2560 -cwiring -P/dev/ttyS0 -b115200 -D -Uflash:w:/tmp/arduino_build_450883/test.ino.hex:i
Reset sent.strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe

This works. But i doen´t now why? :D

TOLDOTECHNIK, 2018-01-17:
We do not have an Arduino MEGA2560 here right now. But it's most likely the _-c_ switch.

_-c arduino is stk500v1_
_-c wiring is stk500v2_

If you look at the boards.txt file in the Arduino hardware folder you will notice the

> mega.menu.cpu.atmega2560.upload.protocol=wiring

line. So in your case you probably need to use the _-c wiring_ switch.

green2blue, 2018-01-17:
Your are right :) with wiring it works 👍 

pi@HHO-Cell:/usr/bin $ **sudo avrdude -D -V -F -c wiring -p m2560 -P /dev/ttyS0 -U flash:w:/tmp/arduino_build_450883/test.ino.hex:i**
Reset sent.
avrdude-original: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude-original: Device signature = 0x1e9801 (probably m2560)
avrdude-original: reading input file "/tmp/arduino_build_450883/test.ino.hex"
avrdude-original: writing flash (2420 bytes):

Writing | ################################################## | 100% 0.37s

avrdude-original: 2420 bytes of flash written

avrdude-original: safemode: Fuses OK (E:FD, H:D8, L:FF)
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe
strace: |avrdude-gpio-autoreset: Broken pipe

avrdude-original done.  Thank you.

strace: |avrdude-gpio-autoreset: Broken pipe
pi@HHO-Cell:/usr/bin $