TOLDOTECHNIK/avrdude-gpio-autoreset #3: Timeout/not responding error

duncanhwood, 2018-06-27:
Hi, I'm having an issue getting this working. Sorry if its a bit of a repeat of the previous issues but I have read and tried the solutions without success.

I'm trying to get this working on a pro mini 5v 328p and a pi a+. I followed the setup instructions and is wired up as per the diagram.
I have checked/tried the following:

- pro mini can be programmed via FTDI/USB using windows.
- serial communication between the arduino and pi is working.
- Try to upload the LED on the arduino flashes, so it looks like the reset is being triggered.
- Tried 0.1uf and 1uf caps.
- Tried different baud rates.
- Tried "-c wiring" and "-c arduino".
- Tried wiping the pi and starting fresh..

Dont know what else to try, here's the output:
`
pi@raspberrypi:~/sketches $ sudo avrdude -D -V -F -c wiring -p m328p -P /dev/ttyAMA0 -U flash:w:/home/pi/sketches/Blink.ino.hex:i -vvv

avrdude-original: Version 6.3
                  Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                  Copyright (c) 2007-2014 Joerg Wunsch

                  System wide configuration file is "/etc/avrdude.conf"
                  User configuration file is "/root/.avrduderc"
                  User configuration file does not exist or is not a regular file, skipping

                  Using Port                    : /dev/ttyAMA0
                  Using Programmer              : wiring
                  Overriding Baud Rate          : 57600
avrdude-original: wiring_open(): releasing DTR/RTS
avrdude-original: wiring_open(): asserting DTR/RTS
Reset sent.avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: ser_recv(): programmer is not responding
avrdude-original: stk500v2_ReceiveMessage(): timeout
avrdude-original: stk500v2_getsync(): timeout communicating with programmer

avrdude-original done.  Thank you.

strace: |avrdude-gpio-autoreset: Broken pipe
`
Any ideas??

TOLDOTECHNIK, 2018-06-28:
Does not sound good. First of all we do have to be sure the serial communication is really working correctly.
What kind of test did you do when you say:

> serial communication between the arduino and pi is working.

We don't quite understand your description on this point:

> Try to upload the LED on the arduino flashes, so it looks like the reset is being triggered.

Can you ensure that the reset on pin GPIO 4 is triggering? Try it with an oscilloscope or attach an LED between GND and GPIO 4. It should briefly flash on reset.

duncanhwood, 2018-06-28:
I uploaded a sketch (using ftdi/windows) that returns the ascii code received from minicom on the pi. ([link](https://oscarliang.com/raspberry-pi-and-arduino-connected-serial-gpio/)) so its definitely sending/receiving.

I just attached an led to make sure the reset is being sent and it does flash. (also tried changing the reset pin on the pi, recompiled, which also makes the led flash)

TOLDOTECHNIK, 2018-06-28:
Did you compile the Blink.ino.hex for the right platform? Can you determine the microcontroller on the board? Is it the ATmega328P?
Please keep us informed about your status. Hope we have some Pro Mini in stock. If so we will test it for you as soon as possible.

duncanhwood, 2018-06-30:
Yes, its compiled for the right platform. 
I've just completely swapped hardware: a pi zero (fresh raspbian install) and a different pro mini 328p 5v 16MHz and am getting the same result.
Again the serial comms are working and reset being sent. 
It's strange as in the previous issue someone was using a pi zero with a pro mini (although the 3.3v version).

I give up. 
Thanks anyway.

TOLDOTECHNIK, 2018-06-30:
Cool! Thanks for your feedback. Patience is worth it 😉

closed 2018-06-30 by TOLDOTECHNIK