TOLDOTECHNIK/avrdude-gpio-autoreset #5: Just another easier way to upload

rin67630, 2021-01-23:
Upload sketches to Arduino

After compiling your own sketch in the Arduino IDE you have to find the generated HEX file and copy it to the Raspberry Pi. On Windows it is located under the temp folder (Press Windows-R, type in %temp% and press Enter). There you will find a build folder containing the HEX file).

Why so clumsy?
Just set the IDE to /dev/usbAMA0 and upload...

TOLDOTECHNIK, 2021-01-26:
Good remarks. Thanks.
Our implementation is mainly intended for automated upload of ready-made firmwares. E.g. for mass production.

rin67630, 2021-01-26:
By the way, if you set a reset pulse by software, the capacitor is not required. 
It was originally useful since the DSR line kept low during the full transmission.

closed 2021-08-05 by TOLDOTECHNIK